LeiWang1999/ZYNQ-NVDLA #1: issue build UMD in MPSoC zynqMP

dovanhuong, 2021-05-31:
hi @LeiWang1999 , I followed your instruction to build UMD for zynqMP, unfortunately, when I ported into zcu102 board, it appeared error as below: 
root@arm:/umd/out/apps/runtime/nvdla_runtime# ./nvdla_runtime
bash: ./nvdla_runtime: cannot execute binary file: Exec format error
root@arm:/umd/out/apps/runtime/nvdla_runtime#
If you have any suggestion, please share to me!
Thank you so much~

LeiWang1999, 2021-05-31:
@dovanhuong libjpeg.a which under external folder should be built by yourself yet. you should build libjpeg9b.

dovanhuong, 2021-05-31:
@LeiWang1999  do you mean I will re-build libjpeg.a? do you have the open source to build libjpeg9b? last time i just $sudo apt install libjpeg9b :((

LeiWang1999, 2021-05-31:
@dovanhuong I think you should find the libjpeg.a under the lib you install by apt and replace the one under external

And, yes libjpeg is opensource and easy to build by oneself

dovanhuong, 2021-05-31:
@LeiWang1999 Yesss expert. However, it have some errors when I compiled your repo as below: could you please take a look and help me some advice~
tony@dovanhuong:~/ZYNQ-NVDLA/umd$ make runtime TOOLCHAIN_PREFIX=/usr/bin/

for dir in core/src/runtime apps/runtime; do \
	make -C $dir; \
done
make[1]: Entering directory '/home/tony/ZYNQ-NVDLA/umd/core/src/runtime'
MODULE_CC /usr/bin/gcc
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/tony/ZYNQ-NVDLA/umd/core/src/runtime'
make[1]: Entering directory '/home/tony/ZYNQ-NVDLA/umd/apps/runtime'
MODULE_CC /usr/bin/gcc
compiling DlaImageUtils.cpp
/usr/bin/gcc  -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration -g -DNVDLA_UTILS_ERROR_TAG="\"DLA_TEST\""  -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I/home/tony/ZYNQ-NVDLA/umd/external/libjpeg-turbo -I./include -I.  -c DlaImageUtils.cpp -MD -MT /home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/DlaImageUtils.o -MF /home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/DlaImageUtils.d -o /home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/DlaImageUtils.o
DlaImageUtils.cpp: In function ‘NvDlaError JPEG2DIMG(std::__cxx11::string, NvDlaImage*, nvdla::IRuntime::NvDlaTensor*)’:
DlaImageUtils.cpp:196:33: error: cannot convert ‘bool’ to ‘boolean’ for argument ‘2’ to ‘int jpeg_read_header(j_decompress_ptr, boolean)’
     jpeg_read_header(&info, true);
                                 ^
/home/tony/ZYNQ-NVDLA/umd/make/compile.mk:61: recipe for target '/home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/DlaImageUtils.o' failed
make[1]: *** [/home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/DlaImageUtils.o] Error 1
make[1]: Leaving directory '/home/tony/ZYNQ-NVDLA/umd/apps/runtime'
Makefile:43: recipe for target 'runtime' failed
make: *** [runtime] Error 2

LeiWang1999, 2021-05-31:
.. I didnot have that issue

LeiWang1999, 2021-05-31:
And for zynq MPSoc， I think you can use the offical prebuilt executable,

LeiWang1999, 2021-05-31:
https://github.com/nvdla/sw/tree/master/prebuilt/arm64-linux

dovanhuong, 2021-05-31:
@LeiWang1999 thank for your recommendation. I used to test with pre-built as you suggested. It has some error as below expert: 
root@smalldla:/arm64-linux# ls
aarch64_nvdla.lua    drm.ko               efi-virtio.rom       images               libnvdla_runtime.so  nvdla_runtime        opendla_1.ko         opendla_2.ko
root@smalldla:/arm64-linux# ./nvdla_runtime 
./nvdla_runtime: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
root@smalldla:/arm64-linux# 

If you have any advice or recommendation, I'm willing to hear from you~
Thank you so much and wish you will have a great thesis defense, cheer~~~

dovanhuong, 2021-05-31:
@LeiWang1999 I completed fix error build with your repo, unfortunately. When I execute that file in board, it still have error as the first one :(((( 
tony@dovanhuong:~/ZYNQ-NVDLA/umd$ make runtime TOOLCHAIN_PREFIX=/usr/bin/

for dir in core/src/runtime apps/runtime; do \
	make -C $dir; \
done
make[1]: Entering directory '/home/tony/ZYNQ-NVDLA/umd/core/src/runtime'
MODULE_CC /usr/bin/gcc
compiling /home/tony/ZYNQ-NVDLA/umd/core/src/common/ErrorLogging.c
/usr/bin/gcc -Os -g -fPIC -finline -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration --std=c99 -DNVDLA_UTILS_ERROR_TAG="\"DLA_RUNTIME\""   -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/core/src/common/include -I/home/tony/ZYNQ-NVDLA/umd/port/linux/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I./include -I. -c /home/tony/ZYNQ-NVDLA/umd/core/src/common/ErrorLogging.c -MD -MT /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/ErrorLogging.o -MF /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/ErrorLogging.d -o /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/ErrorLogging.o
compiling /home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla.c
/usr/bin/gcc -Os -g -fPIC -finline -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration --std=c99 -DNVDLA_UTILS_ERROR_TAG="\"DLA_RUNTIME\""   -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/core/src/common/include -I/home/tony/ZYNQ-NVDLA/umd/port/linux/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I./include -I. -c /home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla.c -MD -MT /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla.o -MF /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla.d -o /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla.o
compiling /home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla_os.c
/usr/bin/gcc -Os -g -fPIC -finline -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration --std=c99 -DNVDLA_UTILS_ERROR_TAG="\"DLA_RUNTIME\""   -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/core/src/common/include -I/home/tony/ZYNQ-NVDLA/umd/port/linux/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I./include -I. -c /home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla_os.c -MD -MT /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla_os.o -MF /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla_os.d -o /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/port/linux/nvdla_os.o
compiling /home/tony/ZYNQ-NVDLA/umd/core/src/common/Check.cpp
/usr/bin/gcc -Os -g -fPIC -finline -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration --std=c++11 -fexceptions -fno-rtti -DNVDLA_UTILS_ERROR_TAG="\"DLA_RUNTIME\""  -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/core/src/common/include -I/home/tony/ZYNQ-NVDLA/umd/port/linux/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I./include -I.  -c /home/tony/ZYNQ-NVDLA/umd/core/src/common/Check.cpp -MD -MT /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/Check.o -MF /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/Check.d -o /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/Check.o
compiling /home/tony/ZYNQ-NVDLA/umd/core/src/common/EMUInterface.cpp
/usr/bin/gcc -Os -g -fPIC -finline -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration --std=c++11 -fexceptions -fno-rtti -DNVDLA_UTILS_ERROR_TAG="\"DLA_RUNTIME\""  -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/core/src/common/include -I/home/tony/ZYNQ-NVDLA/umd/port/linux/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I./include -I.  -c /home/tony/ZYNQ-NVDLA/umd/core/src/common/EMUInterface.cpp -MD -MT /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/EMUInterface.o -MF /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/EMUInterface.d -o /home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime//home/tony/ZYNQ-NVDLA/umd/core/src/common/EMUInterface.o
compiling /home/tony/ZYNQ-NVDLA/umd/core/src/common/EMUInterfaceA.cpp
/usr/bin/gcc -Os -g -fPIC -finline -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Werror-implicit-function-declaration --std=c++11 -fexceptions -fno-rtti -DNVDLA_UTILS_ERROR_TAG="\"DLA_RUNTIME\""  -I/home/tony/ZYNQ-NVDLA/umd/include -I/home/tony/ZYNQ-NVDLA/umd/core/include -I/home/tony/ZYNQ-NVDLA/umd/core/src/common/include -I/home/tony/ZYNQ-NVDLA/umd/port/linux/include -I/home/tony/ZYNQ-NVDLA/umd/external/include -I [... cut ...]
/usr/bin/g++ /home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/nvdla_runtime.mod.o -pthread -L/home/tony/ZYNQ-NVDLA/umd/external/ -ljpeg -L/home/tony/ZYNQ-NVDLA/umd/out/core/src/runtime/libnvdla_runtime -lnvdla_runtime -o /home/tony/ZYNQ-NVDLA/umd/out/apps/runtime/nvdla_runtime/nvdla_runtime -Wl,-rpath=.
make[1]: Leaving directory '/home/tony/ZYNQ-NVDLA/umd/apps/runtime'
tony@dovanhuong:~/ZYNQ-NVDLA/umd$ 

Error in board: 
root@smalldla:/umd/out/apps/runtime/nvdla_runtime# ./nvdla_runtime 
-sh: ./nvdla_runtime: cannot execute binary file: Exec format error
root@smalldla:/umd/out/apps/runtime/nvdla_runtime# 

It is quite weired and confuse for me :(((

LeiWang1999, 2021-05-31:
https://github.com/nvdla/sw/search?q=cannot+execute+binary+file&type=issues

LeiWang1999, 2021-05-31:
Did you execute it on zynq MPSoc?

dovanhuong, 2021-05-31:
yesss bro @LeiWang1999 ! I tried to execute in zcu102 board, the output like that :((( quite strange for me

LeiWang1999, 2021-05-31:
.... it seems that you build it on your local machine but execute it on zynq , it is incorrect

dovanhuong, 2021-05-31:
oh, so what is the correct way bro ?

LeiWang1999, 2021-05-31:
make runtime all in arm.

closed 2021-05-31 by LeiWang1999

dovanhuong, 2021-05-31:
please wait bro~~~ @LeiWang1999 !! it have a problem when I make inside arm board, it appeared not have the make bash, could you please share to me how can you build the make inside zynq enviroment? thank you so much~~~

LeiWang1999, 2021-05-31:
@dovanhuong  you should tranplant a Linux os on your zynq board.
Xilinx Petalinux is recommended.
And instruction was written in my blog...

dovanhuong, 2021-05-31:
hello bro @LeiWang1999 ! I used translated to english from your blog and follow Ubuntu generated by you. Unfortunately, when I make it nothing happened right after as log below:
ubuntu@arm:/umd$ ls
apps  external            log.error  Makefile  utils
core  jpegsrc.v9d.tar.gz  make       port
ubuntu@arm:/umd$ export TOP=${PWD}
ubuntu@arm:/umd$ make runtime TOOLCHAIN_PREFIX=/usr/bin/
ubuntu@arm:/umd$ ls
apps  external            log.error  Makefile  utils
core  jpegsrc.v9d.tar.gz  make       port
ubuntu@arm:/umd$ 
 :((( therefore, I don't know which part I missed! warmly thank you so much~~