LeiWang1999/ZYNQ-NVDLA #2: segmentation fault when running model in zynq fpga

dovanhuong, 2021-06-02:
@LeiWang1999 ! thank for your sharing, I completed to built and run zynq-nvdla in zcu102. However, when I execute model lenet in board it appeared an erro for segmentation as below:
root@nvdla_fpga:~/umd/out/apps/runtime/nvdla_runtime# ./nvdla_runtime --loadable fast-math.nvdla --image ./Images/0_7.jpg --rawdump
creating new runtime context...
Emulator starting
dlaimg height: 28 x 28 x 1: LS: 224 SS: 0 Size: 6272
submitting tasks...
Segmentation fault
root@nvdla_fpga:~/umd/out/apps/runtime/nvdla_runtime# 

This is the insmod when I loaded kmd:
root@nvdla_fpga:~# insmod /lib/modules/4.19.0-xilinx-v2019.1/extra/opendla.ko 
[   38.075678] Probe NVDLA config nvidia,nv_small
[   38.080405] 0 . 12 . 5
[   38.082769] reset engine done
[   38.086171] [drm] Initialized nvdla 0.0.0 20171017 for a0000000.NV_nvdla_wrapper on minor 1
[   38.094544] ------------[ cut here ]------------
[   38.099160] memremap attempted on ram 0x0000000040000000 size: 0x40000000
[   38.105961] WARNING: CPU: 1 PID: 2266 at kernel/iomem.c:108 memremap+0x178/0x1e8
[   38.113341] Modules linked in: opendla(O+) mali(O) uio_pdrv_genirq
[   38.119517] CPU: 1 PID: 2266 Comm: insmod Tainted: G           O      4.19.0-xilinx-v2019.1 #1
[   38.128116] Hardware name: xlnx,zynqmp (DT)
[   38.132285] pstate: 60000005 (nZCv daif -PAN -UAO)
[   38.137059] pc : memremap+0x178/0x1e8
[   38.140705] lr : memremap+0x178/0x1e8
[   38.144349] sp : ffffff800cfb38c0
[   38.147648] x29: ffffff800cfb38c0 x28: ffffff800cfb5000 
[   38.152952] x27: 0000000000000100 x26: ffffff8000b85a80 
[   38.158256] x25: ffffff800cfb3980 x24: 0000000040000000 
[   38.163551] x23: 0000000000000001 x22: ffffffc87985fa18 
[   38.168846] x21: 0000000040000000 x20: 0000000000000004 
[   38.174141] x19: 0000000040000000 x18: 0000000000000010 
[   38.179437] x17: 0000000000000000 x16: 0000000000000000 
[   38.184732] x15: ffffffffffffffff x14: ffffff8009138648 
[   38.190028] x13: ffffff80891d7ecf x12: ffffff80091d7ed8 
[   38.195323] x11: ffffff800914a000 x10: ffffff800cfb35a0 
[   38.200618] x9 : 00000000ffffffd0 x8 : ffffff8008563fe0 
[   38.205914] x7 : 30203a657a697320 x6 : 000000000000015c 
[   38.211209] x5 : 0000000000000007 x4 : 0000000000000000 
[   38.216504] x3 : 0000000000000000 x2 : ffffffffffffffff 
[   38.221800] x1 : c29b4d7389105500 x0 : 0000000000000000 
[   38.227095] Call trace:
[   38.229528]  memremap+0x178/0x1e8
[   38.232829]  dma_init_coherent_memory+0x50/0x110
[   38.237436]  dma_declare_coherent_memory+0x44/0xb8
[   38.242223]  nvdla_drm_probe+0x74/0x98 [opendla]
[   38.246836]  nvdla_probe+0x18c/0x1b0 [opendla]
[   38.251266]  platform_drv_probe+0x50/0xa0
[   38.255266]  really_probe+0x1c8/0x280
[   38.258912]  driver_probe_device+0x54/0xe8
[   38.262992]  __driver_attach+0xe4/0xe8
[   38.266727]  bus_for_each_dev+0x70/0xc0
[   38.270553]  driver_attach+0x20/0x28
[   38.274112]  bus_add_driver+0x1dc/0x208
[   38.277932]  driver_register+0x60/0x110
[   38.281752]  __platform_driver_register+0x44/0x50
[   38.286448]  nvdla_driver_init+0x1c/0x1000 [opendla]
[   38.291397]  do_one_initcall+0x74/0x178
[   38.295216]  do_init_module+0x54/0x1c8
[   38.298948]  load_module+0x1b5c/0x20e0
[   38.302681]  __se_sys_finit_module+0xb8/0xc8
[   38.306934]  __arm64_sys_finit_module+0x18/0x20
[   38.311450]  el0_svc_common+0x84/0xd8
[   38.315095]  el0_svc_handler+0x68/0x80
[   38.318827]  el0_svc+0x8/0xc
[   38.321691] ---[ end trace bf498404d6f095c3 ]---
root@nvdla_fpga:~# 

based on your expertise, do you know any approach to solve this issue? 
Warmly thank you so much,
Tony Do

LeiWang1999, 2021-06-02:
for runtime application, I use clion for remote debug. I think you can use this way to find Segmentation fault point.

And, for opendla.ko. I do not have there too much log... Did you find the NVDLA interruptted and drm driver ? 

You can make a pull request for zcu 102 kmd code for this repo!

dovanhuong, 2021-06-02:
@LeiWang1999 thank for your recommendation. The log as below is the log interrupt:
root@nvdla_fpga:~# cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
  3:       3858       2459       1685       1720     GICv2  30 Level     arch_timer
  6:          0          0          0          0     GICv2  67 Level     zynqmp_ipi
  7:          0          0          0          0     GICv2 175 Level     arm-pmu
  8:          0          0          0          0     GICv2 176 Level     arm-pmu
  9:          0          0          0          0     GICv2 177 Level     arm-pmu
 10:          0          0          0          0     GICv2 178 Level     arm-pmu
 12:          0          0          0          0     GICv2 156 Level     zynqmp-dma
 13:          0          0          0          0     GICv2 157 Level     zynqmp-dma
 14:          0          0          0          0     GICv2 158 Level     zynqmp-dma
 15:          0          0          0          0     GICv2 159 Level     zynqmp-dma
 16:          0          0          0          0     GICv2 160 Level     zynqmp-dma
 17:          0          0          0          0     GICv2 161 Level     zynqmp-dma
 18:          0          0          0          0     GICv2 162 Level     zynqmp-dma
 19:          0          0          0          0     GICv2 163 Level     zynqmp-dma
 20:          0          0          0          0     GICv2 164 Level     Mali_GP_MMU, Mali_GP, Mali_PP0_MMU, Mali_PP0, Mali_PP1_MMU, Mali_PP1
 21:          0          0          0          0     GICv2 109 Level     zynqmp-dma
 22:          0          0          0          0     GICv2 110 Level     zynqmp-dma
 23:          0          0          0          0     GICv2 111 Level     zynqmp-dma
 24:          0          0          0          0     GICv2 112 Level     zynqmp-dma
 25:          0          0          0          0     GICv2 113 Level     zynqmp-dma
 26:          0          0          0          0     GICv2 114 Level     zynqmp-dma
 27:          0          0          0          0     GICv2 115 Level     zynqmp-dma
 28:          0          0          0          0     GICv2 116 Level     zynqmp-dma
 32:          0          0          0          0     GICv2  49 Level     cdns-i2c
 33:          0          0          0          0     GICv2  50 Level     cdns-i2c
 34:          0          0          0          0     GICv2  42 Level     ff960000.memory-controller
 35:          0          0          0          0     GICv2  57 Level     axi-pmon, axi-pmon
 36:          0          0          0          0     GICv2 155 Level     axi-pmon, axi-pmon
 37:          0          0          0          0     GICv2 150 Level     nwl_pcie:misc
 42:         45          0          0          0     GICv2  47 Level     ff0f0000.spi
 43:          0          0          0          0     GICv2  58 Level     ffa60000.rtc
 44:          0          0          0          0     GICv2  59 Level     ffa60000.rtc
 45:          0          0          0          0     GICv2 165 Level     ahci-ceva[fd0c0000.ahci]
 46:       1730          0          0          0     GICv2  81 Level     mmc0
 47:        121          0          0          0     GICv2  53 Level     xuartps
 50:          0          0          0          0     GICv2  84 Edge      ff150000.watchdog
 51:          0          0          0          0     GICv2  88 Level     ams-irq
 52:          0          0          0          0     GICv2 154 Level     fd4c0000.dma
 53:          0          0          0          0     GICv2 151 Level     fd4a0000.zynqmp-display
 54:          0          0          0          0     GICv2 121 Level     a0000000.NV_nvdla_wrapper
 56:          0          0          0          0     GICv2 101 Level     dwc3-otg
IPI0:      1505       1395        999       1595       Rescheduling interrupts
IPI1:        20        222        159        190       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
IPI4:       174        382        394        405       Timer broadcast interrupts
IPI5:         0          0          0          0       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts
Err:          0
root@nvdla_fpga:~# ls /dev/dri/
by-path  card0  card1  renderD128

I'll make a pull request for zcu102 as you said!
thank you so much~

LeiWang1999, 2021-07-21:
Hi, @dovanhuong 
Could you tell me which revision of ZCU 102 you used to map NVDLA ?
rev 1.0 or rev 1.1 ?.

dovanhuong, 2021-07-21:
hi @LeiWang1999 , 
I used ZCU102 rev 1.1. As regard to this issue I fixed that by setup the memory map again in the tree device, It's working good under nvdla small. I'm working in build nvdla full configuration. If you also working with this, please share to me and we can cooperate. Hope this can help you~~
Cheer~

LeiWang1999, 2021-07-21:
@dovanhuong Thanks.

And now I am working on mapping small nvdla to ZCU 102 rev 1.1 board, but it seems that there are some unnoticed problem with this board and now it trapped me for few days.

I got some mmap issue when i compile the Runtime and when execute with loadable , and I doubt it was caused by this board.  But, there are too many people working with rev 1.0 rather than 1.1, so it is very lucky for me to find a person who works with same platform.

Did you have some contact information? Like Wechat、Telergam？

dovanhuong, 2021-07-21:
@LeiWang1999 , if you don't mind send me your error via my email: vanhuong.do@asicland.com or add my Telegram by phone number: +821066292899

closed 2021-07-24 by LeiWang1999