LeiWang1999/ZYNQ-NVDLA #3: NVDLA Runtime error

qaziullah, 2021-07-01:
Hi @LeiWang1999, I am trying to run NVDLA on FPGA and having  some issues. I was hopping if you could guide me in this regard.

I have implemented nvdla small on Intel FPGA (ARRIA10) with 32bit HPS (ARM A 9). I have modified kmd and umd to make it compatible with 32bit arm processor. 
I am able to successfully the module but while running the runtime test of PDP I am having the following error messages.  
[image]

and for the other test the kernel got stuck in the middle and become unresponsive. 

Note: 
I am using kernel 4.14, 512 MBs are reserved for KMD. 
Could you kinldy look into the issue and let me know root cause of the errors. 

Thanks in advance.

LeiWang1999, 2021-07-02:
You may checkout wheather the system interrupt or the drm device had been loaded correctly

```
root@arm:~# insmod /lib/modules/4.19.0-xilinx-v2019.1/extra/opendla.ko 
root@arm:~# cat /proc/interrupts 
           CPU0       CPU1       
 16:          1          0     GIC-0  27 Edge      gt
 17:          0          0     GIC-0  43 Level     ttc_clockevent
 18:       4106       3966     GIC-0  29 Edge      twd
 19:          0          0     GIC-0  37 Level     arm-pmu
 20:          0          0     GIC-0  38 Level     arm-pmu
 21:         43          0     GIC-0  39 Level     f8007100.adc
 24:          0          0     GIC-0  35 Level     f800c000.ocmc
 25:        291          0     GIC-0  59 Level     xuartps
 26:         16          0     GIC-0  51 Level     e000d000.spi
 27:        427          0     GIC-0  54 Level     eth0
 28:       4915          0     GIC-0  56 Level     mmc0
 29:          0          0     GIC-0  45 Level     f8003000.dmac
 30:          0          0     GIC-0  46 Level     f8003000.dmac
 31:          0          0     GIC-0  47 Level     f8003000.dmac
 32:          0          0     GIC-0  48 Level     f8003000.dmac
 33:          0          0     GIC-0  49 Level     f8003000.dmac
 34:          0          0     GIC-0  72 Level     f8003000.dmac
 35:          0          0     GIC-0  73 Level     f8003000.dmac
 36:          0          0     GIC-0  74 Level     f8003000.dmac
 37:          0          0     GIC-0  75 Level     f8003000.dmac
 38:          0          0     GIC-0  40 Level     f8007000.devcfg
 44:          0          0     GIC-0  41 Edge      f8005000.watchdog
 45:          0          0     GIC-0  61 Level     40000000.NV_nvdla_wrapper
IPI1:          0          0  Timer broadcast interrupts
IPI2:       1267       2841  Rescheduling interrupts
IPI3:          4          3  Function call interrupts
IPI4:          0          0  CPU stop interrupts
IPI5:          0          0  IRQ work interrupts
IPI6:          0          0  completion interrupts
```

qaziullah, 2021-07-02:
Thanks @LeiWang1999  for the reply. 
 
I have checked the interrupts and yes system interrupt has been loaded. Here is the putty log.
[image]

I believe that drm device is also loaded. Here is the content of /dev/dri.
[image]

LeiWang1999, 2021-07-02:
@qaziullah I didn't try the loadable given by official repo, and I am worry so.
You can the loadable I had put on here. -> https://github.com/LeiWang1999/nvdla_loadables

qaziullah, 2021-07-05:
@LeiWang1999  I tried using the loadables given by you, bu I got error saying "Unsupported color conversion request". 
This error was also mentioned [here ](https://github.com/nvdla/sw/issues/164)but with no solution. Could you look into it.

LeiWang1999, 2021-07-08:
@qaziullah Yeah, Sorrry for the late reply. Actually, I got this issue, and I solved it.

And the solution can be found in my graduation paper,  BGR2RGB operation in this source file `info.out_color_space = JCS_EXT_BGR;  `  should be commented .

[image]

qaziullah, 2021-07-08:
Hi @LeiWang1999, Thanks for solution. It worked and now there is no Color Conversion error prompt. 

But when I ran the runtime on the loadables provide by you [here](https://github.com/LeiWang1999/nvdla_loadables) I am getting some weird long logs on terminal ending with test passed remarks.
The output for the command ```./nvdla_runtime --loadable resnet18-cifar10-caffe/loadables/basic.nvdla --image resnet18-cifar10-caffe/Image/cat_32.jpg```. 

[image]
Is it expected output or am I doing something wrong? 
Kindly look into it and let me know.

qaziullah, 2021-07-08:
@LeiWang1999. I got it. I wasn't using --rawdump option. So I wasn't able to read the output.dimg. 

Finally It worked. Thank You @LeiWang1999 very much. I was stuck on the project for a while, but thanks to your help I was able to make nvdla successfully on FPGA. 
Once Again Thank you very much.

LeiWang1999, 2021-07-08:
@qaziullah well done,  it worked. to see the output you should append additional argument —rawdump and cat the output.dimg file

qaziullah, 2021-07-08:
@LeiWang1999, Okay I got it. 
Thanks again.

closed 2021-07-08 by qaziullah

LeiWang1999, 2021-07-08:
give me star XD

qaziullah, 2021-07-08:
I already gave it.

nagendra7890, 2021-09-14:
hello @qaziullah 
I was trying to run Lenet with mnist but I was not able to see the execution time and test result. there was no response it got stuck.
please see this  https://github.com/nvdla/sw/issues/177#issuecomment-913763834 https://github.com/nvdla/sw/issues/177#issuecomment-917419962

qaziullah, 2021-09-15:
@nagendra7890 I was having the same problem. I got it resolved by defining all the signal of AXI interface (axi 4 used for interfacing external RAM) and defining the interrupt signal type as active high ( at first the signal type of the interrupt was set to active low by default in the IP packager).  I had done both of these changes simultaneously so I can't exactly say which one of the two made it work, or maybe both of them. 
Also try to reserve a good portion of RAM ( in device tree file) for NVDLA_IP to interface with. 

@LeiWang1999 has done good work in defining all the AXI signal according to the AXI interface standard documentation. Try to use the top file from @LeiWang1999's github directory where all the AXI signals are defined.

This solution worked for me and I hope that it'll work for you.

LeiWang1999, 2021-09-16:
@qaziullah Did you see my recent work, we can use nvdla to inference yolox now!

qaziullah, 2021-09-16:
@LeiWang1999  yes I have seen it, great work from you.  I have been little busy recently so Haven't tried it yet. Am gonna try it soon. 

Also I am planning to port the other hw versions (nv_small_256 MAC, nv_medium_512 and others) to FPGA and test performance difference between them. To test the performance I was thinking about about doing the continuous inference on collection of images. Have you tried doing inference on batch of Images at once. Because at  nvdla_runtime is doing inference on single image at once and loading the dla loadable is expensive in term of time. as mentioned [here](https://github.com/nvdla/sw/issues/38#issue-319323289). Have done something in this regard?

LeiWang1999, 2021-09-16:
@qaziullah  Use the official  nvdla_runtime to inference batch images is not a simple work, you should customize runtime. However, my recent work about a new toolchain for nvdla can do that in an easy way. Maybe this can help you.

And the implementation of different hw versions on FPGA is a good work. I wanna to do so but I did not have a board with enough lut.

qaziullah, 2021-09-16:
Thanks @LeiWang1999. I'll look into the new tool chain.

About the different Configuration of NVDLA, I have Compiled the hardware model of (nv_small, nv_small_256 and nv_medium_512) and lucky my FPGA is able to fit all of them (SW stack for the configurations other nv_small has to be verified. Maybe the new tool chain you have mentioned above will help me with the SW part). 
Soon I'll update project status and share it on github.

nagendra7890, 2021-09-19:
Hi, @qaziullah I have tried @LeiWang1999 RTL but still facing the same issue. I don't know what's wrong! I used AXI-smart connect do I need to change the vivado version? currently, I am using vivado,petalinux 2018.3