LeiWang1999/ZYNQ-NVDLA #4: gcc: internal compiler error: Killed (program cc1plus)

sunny-yellow, 2021-08-24:
gcc: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
/ZYNQ-NVDLA-master/umd/make/compile.mk:61: recipe for target '/ZYNQ-NVDLA-master/umd/out/core/src/compiler/libnvdla_compiler/engine-ast/FullyConnectedOp.o' failed
make[1]: *** [/ZYNQ-NVDLA-master/umd/out/core/src/compiler/libnvdla_compiler/engine-ast/FullyConnectedOp.o] Error 4
[ 2068.232790] Out of memory: Kill process 14355 (cc1plus) score 52 or sacrifice child
[ 2068.240476] Killed process 14355 (cc1plus) total-vm:78112kB, anon-rss:52848kB, file-rss:0kB, shmem-rss:0kB
我在编译compiler时执行如下命令
$ make -j `nproc` TOP=${PWD} TOOLCHAIN_PREFIX=/usr/bin/ compile您
出现了这个错误，一定要靠交换分区来解决吗，不知你遇到了没有

sunny-yellow, 2021-08-24:
@LeiWang1999

LeiWang1999, 2021-08-24:
@sunny-yellow 我没有遇到过这个错误，但看起来是因为你在做文件系统的时候么有给swap分配足够的空间？
可以参考该文章构建文件系统：
https://zhuanlan.zhihu.com/p/392974835

sunny-yellow, 2021-08-25:
> @sunny-yellow 我没有遇到过这个错误，但看起来是因为你在做文件系统的时候么有给swap分配足够的空间？
> 可以参考该文章构建文件系统：
> https://zhuanlan.zhihu.com/p/392974835

好的，那该分多大的swap空间合适呢

LeiWang1999, 2021-08-25:
@sunny-yellow 我是直接用默认的（
1. 方便提供一下您的器件型号吗
2. 直接编runtime可行吗？

sunny-yellow, 2021-08-25:
zcu104
我分了2个G的swap，compiler时超级慢

sunny-yellow, 2021-08-25:
> @sunny-yellow 我是直接用默认的（
> 
> 1. 方便提供一下您的器件型号吗
> 2. 直接编runtime可行吗？

好了好了编译出来了
还有个问题就是您博客里说的那个
$ cp ~/ZYNQ-NVDLA/umd/out/core/src/compiler/libnvdla_compiler/libnvdla_compiler.so -r ~/Tengine/source/device/opendla/lib/
这个最后的lib这个文件是要把petalinux里生成的rootfs.tar.gz解压出来的那个lib文件夹吗考到Tengine里吗

LeiWang1999, 2021-08-25:
@sunny-yellow 是的，就是很慢，因为实际上这个是在主机编译，觉得慢的话可以交叉编译（

sunny-yellow, 2021-08-25:
> > @sunny-yellow 我是直接用默认的（
> > 
> > 1. 方便提供一下您的器件型号吗
> > 2. 直接编runtime可行吗？
> 
> 好了好了编译出来了
> 还有个问题就是您博客里说的那个
> $ cp ~/ZYNQ-NVDLA/umd/out/core/src/compiler/libnvdla_compiler/libnvdla_compiler.so -r ~/Tengine/source/device/opendla/lib/
博主这个lib文件官方里没有

sunny-yellow, 2021-08-25:
> @sunny-yellow 是的，就是很慢，因为实际上这个是在主机编译，觉得慢的话可以交叉编译（

好了好了编译出来了
还有个问题就是您博客里说的那个
$ cp ~/ZYNQ-NVDLA/umd/out/core/src/compiler/libnvdla_compiler/libnvdla_compiler.so -r ~/Tengine/source/device/opendla/lib/
博主这个lib文件官方里没有

LeiWang1999, 2021-08-25:
@sunny-yellow 是没有，需要自己新建一个文件夹，或者直接 cp -rf

sunny-yellow, 2021-08-25:
> @sunny-yellow 是没有，需要自己新建一个文件夹，或者直接 cp -rf

OK，谢谢

closed 2021-08-25 by LeiWang1999