LeiWang1999/ZYNQ-NVDLA #5: Linking CXX shared library error

balckwilliam, 2021-08-24:
/usr/bin/ld: /root/Tengine/source/device/opendla/lib/libprotobuf.a(descriptor.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against external symbol `__stack_chk_guard@@GLIBC_2.17' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /root/Tengine/source/device/opendla/lib/libprotobuf.a(descriptor.o)(.text+0x50c): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `__stack_chk_guard@@GLIBC_2.17'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
source/CMakeFiles/tengine-lite.dir/build.make:7192: recipe for target 'source/libtengine-lite.so' failed
make[3]: *** [source/libtengine-lite.so] Error 1
CMakeFiles/Makefile2:325: recipe for target 'source/CMakeFiles/tengine-lite.dir/all' failed
make[2]: *** [source/CMakeFiles/tengine-lite.dir/all] Error 2
CMakeFiles/Makefile2:618: recipe for target 'examples/CMakeFiles/tm_classification_opendla.dir/rule' failed
make[1]: *** [examples/CMakeFiles/tm_classification_opendla.dir/rule] Error 2
Makefile:322: recipe for target 'tm_classification_opendla' failed
make: *** [tm_classification_opendla] Error 2

最后链接libprotobuf.a有问题，大佬遇到过吗？

LeiWang1999, 2021-08-24:
以前有碰到过，当时是参考这个方案解决的：
https://blog.csdn.net/qq_30653631/article/details/107620137

closed 2021-08-25 by LeiWang1999

talahimam, 2021-10-06:
Hi @LeiWang1999,

I am trying to cross-compile Tengine using this cross compiler: https://releases.linaro.org/components/toolchain/binaries/7.3-2018.05/aarch64-linux-gnu/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu.tar.xz. I've cross-compiled the runtime, compiler and protobuf libraries, but when I build Tengine using the cross-compiler, it throws this error in linking the CXX shared library:

**/home/simam/Downloads/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.3.1/../../../../aarch64-linux-gnu/bin/ld: /home/simam/Senior_Proj/Tengine/source/device/opendla/lib/libprotobuf.a(message.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against external symbol `_ZNSt8ios_base4InitD1Ev@@GLIBCXX_3.4' can not be used when making a shared object; recompile with -fPIC
/home/simam/Downloads/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.3.1/../../../../aarch64-linux-gnu/bin/ld: /home/simam/Senior_Proj/Tengine/source/device/opendla/lib/libprotobuf.a(message.o)(.text.startup+0x30): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `_ZNSt8ios_base4InitD1Ev@@GLIBCXX_3.4'
/home/simam/Downloads/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.3.1/../../../../aarch64-linux-gnu/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status**

I went through the blog you mentioned in your comment, but how would you go about fixing this error for a cross compiler?

yss-pku, 2022-02-24:
我也碰到了这个问题，但是用上面这个方法没有解决,还是报同样的错误。

`/usr/bin/ld: /Tengine/source/device/opendla/lib/libprotobuf.a(descriptor.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against external symbol `_ZNSs4_Rep20_S_empty_rep_storageE@@GLIBCXX_3.4' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /Tengine/source/device/opendla/lib/libprotobuf.a(descriptor.o)(.text+0x8f8): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `_ZNSs4_Rep20_S_empty_rep_storageE@@GLIBCXX_3.4'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
source/CMakeFiles/tengine-lite.dir/build.make:6550: recipe for target 'source/libtengine-lite.so' failed
gmake[3]: *** [source/libtengine-lite.so] Error 1
CMakeFiles/Makefile2:103: recipe for target 'source/CMakeFiles/tengine-lite.dir/all' failed
gmake[2]: *** [source/CMakeFiles/tengine-lite.dir/all] Error 2
CMakeFiles/Makefile2:588: recipe for target 'examples/CMakeFiles/tm_classification_opendla.dir/rule' failed
gmake[1]: *** [examples/CMakeFiles/tm_classification_opendla.dir/rule] Error 2
Makefile:279: recipe for target 'tm_classification_opendla' failed
gmake: *** [tm_classification_opendla] Error 2`

我看有人说加上-fPIC, 具体在makefile哪里加呢

LeiWang1999, 2022-02-24:
在编译protobuf的时候加上，不是在umd这个工程的makefile里面加。

yss-pku, 2022-02-27:
> 在编译protobuf的时候加上，不是在umd这个工程的makefile里面加。

感谢，已经解决！

wei8171023, 2022-09-23:
> > 在编译protobuf的时候加上，不是在umd这个工程的makefile里面加。
> 
> 感谢，已经解决！

请问你是怎么解决的呢？在makefile哪里加上-fPIC呢？