LeiWang1999/ZYNQ-NVDLA #6: Package IP Error

liqx-github, 2021-09-06:
Appears when Package IP：
**[IP_Flow 19-3252] Bus Interface 'm_axi': The master AXIMM interface must have required port maps to the read or write channel signals.**
Does anyone know how to solve it？

LeiWang1999, 2021-09-07:
Did you have some more detail about this issue? like a screenshot.

liqx-github, 2021-09-07:
[image]

After automatically inferring the port：

[image]

LeiWang1999, 2021-09-07:
@liqx-github 你package axi协议线的时候，没有吧下面那些 nvdla_core2dbb*的信号线也加上去，这些也是axi总线的一部分。。。

liqx-github, 2021-09-07:
那请问我应该怎么做呢？(第一次弄这个，不太懂)

liqx-github, 2021-09-07:
是需要在自动推断之前设置些什么吗？还是得手动连接？

liqx-github, 2021-09-07:
是这样吗？
[image]
为什么在Memory Map的时候没有包装好的APB总线？
您能否帮忙用个人电脑试一下，看看是不是我自己的vivado版本的问题？
十分感谢！

LeiWang1999, 2021-09-07:
@liqx-github 我试过，没有问题哎，Memory Map本来就是没有apb的，需要自己加一个，看我的tutorial

liqx-github, 2021-09-07:
再请问一下，上面那张图ports and interface那里，两个warning是可以不用管吗？我点自动推断的话就会变成一开始nvdla_core2dbb*的信号线没加上的情况

LeiWang1999, 2021-09-07:
@liqx-github 嗯，是不用管的。
顺便问一下：考虑一下 给这个项目点个star & follow一下作者 吗？

liqx-github, 2021-09-07:
当然

closed 2021-09-13 by LeiWang1999