LeiWang1999/ZYNQ-NVDLA #7: tengine下执行分类报错

sunny-yellow, 2021-10-09:
tengine-lite library version: 1.5-dev
(DLA) Error 0x00000004: (in /root/umd/utils/BuddyAlloc.c, function construct(), line 180)
(DLA) Error 0x00000004: (propagating from Memory.cpp, function init(), line 95)
(DLA) Error 0x00000004: (propagating from engine-ast/EngineGraph.cpp, function initGraphResources(), line 107)
terminate called after throwing an instance of 'nvdla::priv::NvErrorException'
Aborted
博主，这个错你了解吗

tongjiaxuan666, 2022-01-02:
请问这个问题是如何解决的呢？我遇到了同样的问题

tongjiaxuan666, 2022-01-05:
非常期待你的回复

White-Sunday, 2024-01-18:
这是来自nvdla compiler的报错，我已经在我的环境里解决。希望能对后面遇到这个问题的人产生一点帮助。
我的环境：
硬件：Zynq 7000 AX7Z100（处理器：Arm Cortex-A9 ， 32 位的处理器架构， ARMv7-A 指令集体系结构）
操作系统：
```
$ uname -a
Linux arm 4.19.0-xilinx-v2019.1 #1 SMP PREEMPT Sat Dec 16 10:23:29 UTC 2023 armv7l armv7l armv7l GNU/Linux

```
内存：仅有703M的物理内存可用，且系统没有启用虚拟内存。
```
$ free -m -h
              total        used        free      shared  buff/cache   available
Mem:           752M         19M        235M        9.9M        496M        703M
Swap:            0B          0B          0B
```
经过调试，我发现问题在`umd/core/src/compiler/Memory.cpp ` 82-99中的92 `m_base_addr = malloc(poolSize)`，malloc请求分配的空间太大了，内存不够用，返回0，导致后面的检查过不了报错。

解决方法为，尝试重新设置`umd/core/src/compiler/include/priv/Profile.h` 里250 -251的大小为：
```
            m_globalDRAMSize(1LLU << 27),
            m_localDRAMSize(1LLU << 38),
```
即128MB和256MB（原来是500MB和1GB）。

重新编译你的compiler。