LeiWang1999/ZYNQ-NVDLA #9: Build the Tengine with RiscV+NVDLA Project

Darshcg, 2021-11-10:
Hi @LeiWang1999,

Thank you for your work.

Can you please help me with the steps to Build the Tengine with RiscV+NVDLA Project? What are the changes to be made? Would be a great help from your side.

Best Regards,
Darshan C G

Darshcg, 2021-11-11:
Hi @LeiWang1999

Looking forward to your reply.

Thanks,
Darshan C G

LeiWang1999, 2021-11-11:
@Darshcg Sorry for the late reply. The work of Tengine + NVDLA Toolchain were tested with a ZCU102 evaluation board which have an arm A53 processor and some programmable logic, so I don't have such an environment and ideas for the  RiscV+NVDLA solution. But, actually Tengine supports riscv backend and I think this toolchain shouldn't be difficult to be deployed in such a platform.
So, What's troubling you now? I think you just need a linux operating system and it will probably work well.

Darshcg, 2021-11-11:
Hi @LeiWang1999,

Thank you a lot for your kind reply.

I have RiscV+NVDLA working with Linux and I am able to run the Caffe model successfully.

But I want to port Tengine to the present SW stack. 

What I understood is: I had to cross-compile the Compiler and Runtime and Tengine with RiscV toolchain on my X86 Platform and copy the output executable onto the target for Inference.

So, my Doubt is how to cross-compile the Compiler and Tengine with RiscV toolchain?

Best Regards,
Darshan C G

LeiWang1999, 2021-11-11:
@Darshcg Well, I connected the board with internet and I do all the compilation process on the board instead of corss-compile.

Darshcg, 2021-11-11:
Hi @LeiWang1999,

Thank you for your reply.

Is there any way to do it?

I tried to install the tengine for riscv through Cross compilation referring to this(https://github.com/OAID/Tengine/blob/tengine-lite/doc/docs_zh/source_compile/compile_linux.md#%E4%BA%A4%E5%8F%89%E7%BC%96%E8%AF%91-arm3264-linux-%E7%89%88%E6%9C%AC), But I am facing the issue while building.

Assembler messages:
Error: cannot find default versions of the ISA extension `v'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:60: Error: unrecognized opcode `vsetvli t0,a0,e32'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:82: Error: unrecognized opcode `vlw.v v0,(t3)'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:83: Error: unrecognized opcode `vlw.v v1,(t1)'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:87: Error: unrecognized opcode `vsw.v v0,(a2)'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:89: Error: unrecognized opcode `vsw.v v1,(a2)'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:95: Error: unrecognized opcode `vlw.v v0,(t3)'
/home/darshan/edge/Tengine/source/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S:96: Error: unrecognized opcode `vsw.v v0,(a2)'
source/CMakeFiles/tengine-lite.dir/build.make:1921: recipe for target 'source/CMakeFiles/tengine-lite.dir/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S.o' failed
make[2]: *** [source/CMakeFiles/tengine-lite.dir/device/cpu/op/conv/risc-v/lp64dv/im2col_fp32_1x1.S.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'source/CMakeFiles/tengine-lite.dir/all' failed
make[1]: *** [source/CMakeFiles/tengine-lite.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

Can you please help me cross-compiling the Tengine for riscV?

Thanks,
Darshan C G

Darshcg, 2021-11-15:
Hi @LeiWang1999,

I want to run Yolov3-Tiny or Yolov3 on NVDLA small Configuration. I followed this repo https://github.com/prasshantg/darknet, where he split the loadables. But I got to know from this thread(https://github.com/CSL-KU/firesim-nvdla/issues/18) that those were being generated for large configuration. Do we have any approach on how to split the prototxt file and generate the split loadables? And if we generate them for small configuration, can we really run them on NVDLA small?

Looking forward to your reply.

Best Regards,
Darshan C G

Darshcg, 2021-11-17:
Hi @LeiWang1999,

Looking forward to your reply.

Best Regards,
Darshan C G

LeiWang1999, 2021-11-18:
@Darshcg Well, given my preceding research, the loadable file in that darknet-nvdla repository was generated by non-public tool and it's difficult to generate it by ourselves. But I don‘t think that is a good solution because darknet can only inference few models and it seems that the pre-generated loadable is configured with nvdla-large spec.

LeiWang1999, 2021-11-21:
@Darshcg Good news for you, have you ever heard about "zhihu" which is a chinese discussion forum. And I post my technical post about this work on [it](https://zhuanlan.zhihu.com/p/401943271). Today sb left a message under this post shows that he successfully mapped this Tengine toolchain into a rvv system built from chipyard, so I think his solution will be suitable for you, and here is his zhihu's homepage:https://www.zhihu.com/people/mei-ben-shan maybe you can contact and collaborate with him.

closed 2022-02-24 by LeiWang1999

rookie0620, 2023-10-10:
Hi @Darshcg,

I want to do the same thing with you: run yolo  or other net on Boom with NVDLAsmall, do you accomplish this work or what other issues remain unresolved？
Looking forward to your reply.