LeiWang1999/ZYNQ-NVDLA #11: vivado 自动推导axi和apb连线

tongjiaxuan666, 2021-12-07:
[image]
您好，看了您的fpga映射博客，有如下几个疑问：
1.vivado封装IP包装AXI和APB总线的 时候vivado自动推导的意思是进入port界面如果出现了vivado自动识别的总线对应就可以了嘛。
2.包装完总线后是不是需要绑定axi和apb与clock信号 
3.题主提供的constant IP是为了让以太网复位吗？可以不用这个IP吗？
4.address editor 中的地址映射和kmd移植中的设备树地址有关系吗？

LeiWang1999, 2021-12-07:
1.自动推到，比如说你要把AXI信号线都选上，他会自动帮你跟AXI协议对上
2.印象里我试过绑定与没有绑定都可以，最好还是绑一下吧，csb clock-apb core-axi
3.不用没有关系
4.有关系，但是这个步骤在整个workflow中没有影响，因为你后来设备树是直接抄pl.dts的，那边也会自动变

tongjiaxuan666, 2021-12-07:
感谢题主回复，这些天在复现题主博客中的硬件端的移植过程，block design综合完成了，综合，impletation,也做完了，到sdk里跑题主提供的santity,在串口中运行结果是这样的
[image]

我用的板卡型号和题主毕设用的型号一样是xc7z045ffg900-2发现0x20000000和0x30000000的位置并不是DDR3的区域。
[image]
然后我把SDK里的黄金模型放在了0x15000000处，baseaddress放在0x10000000处，通过验证0x10200000和0x15000000处的数据是否能对应上，可是结果对应不上，这是说明硬件系统出问题了吗？

在debug过程中我怀疑在操作过程中的两个点：
1.包装axi和apb的过程里直接进入这个界面好像vivado就出现了这个界面，我默认为它自动推导好了，没有再进行别的改动，这样可以吗：
[image]
2.DDR3大小是板卡型号决定的吧，可是我与题主用的芯片都是zynq的xc7z045，理论上应该没什么不同，是在哪里还需要后续配置吗？

tongjiaxuan666, 2021-12-14:
谢谢题主，之前是操作的一些小问题，现在都解决了

closed 2021-12-14 by tongjiaxuan666

ChangAnGuLi, 2022-05-13:
> 谢谢题主，之前是操作的一些小问题，现在都解决了

您好，我与您同样的是在芯片为zynq的xc7z045上进行的映射工作，但是我在block design阶段遇到了一个问题，请问您可以帮忙看一下吗
[image]
十分感激，谢谢