LeiWang1999/ZYNQ-NVDLA #12: zc706 runtime error because of vmap

tongjiaxuan666, 2021-12-14:
zc706移植nvdla
[image]
检查驱动发现
[image]
申请vmalloc的大小空间太大超过要求。
板卡芯片为zynq7000系列的zc045ffg900-2,kmd用的仓库里的没有更改，为nvdla的dma预留了256MB，为什么会动态内存超出范围呢？题主有遇到类似的问题吗？该从什么方向上调整呢？

LeiWang1999, 2021-12-14:
看起来是读loadable就出问题了，这种情况下我会用clion或者vscode remote来打断点调试看看

tongjiaxuan666, 2021-12-14:
那把给nvdla预留的256M改成128M原理上可行嘛？

closed 2021-12-21 by tongjiaxuan666

Mightyyy, 2022-02-12:
@tongjiaxuan666 hello我也遇到了相同的问题，请问你最后是怎么解决的呢

tongjiaxuan666, 2022-02-12:
> @tongjiaxuan666 hello我也遇到了相同的问题，请问你最后是怎么解决的呢
https://github.com/nvdla/sw/issues/99 是kernel内核版本的问题，内核4.13版本以后，kmd中一个函数的dma逻辑要取反，详细解决方案在这个issue里

Mightyyy, 2022-02-12:
> > @tongjiaxuan666 hello我也遇到了相同的问题，请问你最后是怎么解决的呢
> > [nvdla/sw#99](https://github.com/nvdla/sw/issues/99) 是kernel内核版本的问题，内核4.13版本以后，kmd中一个函数的dma逻辑要取反，详细解决方案在这个issue里

感谢你的回答！但我将if(!dma)改为了if(dma)，问题似乎并没有解决，并且还多了failed to register drm device这个问题，请问你有遇到这个情况吗

[   44.446857] [drm] Initialized nvdla 0.0.0 20171017 for 40000000.NV_nvdla_wrapper on minor 0
[   44.456561] vmap allocation for size 268439552 failed: use vmalloc=<size> to increase size
[   44.456640] NVDLA 40000000.NV_nvdla_wrapper: failed to register drm device
[   44.462714] NVDLA: probe of 40000000.NV_nvdla_wrapper failed with error -12

tongjiaxuan666, 2022-02-12:
> > > @tongjiaxuan666 hello我也遇到了相同的问题，请问你最后是怎么解决的呢
> > > [nvdla/sw#99](https://github.com/nvdla/sw/issues/99) 是kernel内核版本的问题，内核4.13版本以后，kmd中一个函数的dma逻辑要取反，详细解决方案在这个issue里
> 
> @tongjiaxuan666 感谢你的回答！但我将if(!dma)改为了if(dma)，问题似乎并没有解决，并且还多了failed to register drm device这个问题，请问你有遇到这个情况吗
> 
> [ 44.446857] [drm] Initialized nvdla 0.0.0 20171017 for 40000000.NV_nvdla_wrapper on minor 0 [ 44.456561] vmap allocation for size 268439552 failed: use vmalloc= to increase size [ 44.456640] NVDLA 40000000.NV_nvdla_wrapper: failed to register drm device [ 44.462714] NVDLA: probe of 40000000.NV_nvdla_wrapper failed with error -12

[image]
还要检查一下这个地方有没有修改，图片是王维大佬的博客，作者的参考文献下有

Mightyyy, 2022-02-13:
> [image] 还要检查一下这个地方有没有修改，图片是王维大佬的博客，作者的参考文献下有

@tongjiaxuan666 谢谢大佬，我又试了一下好像还是不行，我取反了dma并改了函数名还是上面那三个问题。请问你用的是什么版本的Linux kernel啊？可以加微信私聊一下嘛 :)

tongjiaxuan666, 2022-02-13:
> @tongjiaxuan666 谢谢大佬，我又试了一下好像还是不行，我取反了dma并改了函数名还是上面那三个问题。请问你用的是什么版本的Linux kernel啊？可以加微信私聊一下嘛 :)

wx:tjx18845636178 
内核版本是4.19