LeiWang1999/ZYNQ-NVDLA #14: Module not found when generating bitstream

Mightyyy, 2022-02-08:
Hi, I am facing the following problem when generating bitstream for ZC706:
[image]
It shows that [Synth 8-439] module 'CKLNQD12' not found.
Please help me solve these errors. Thanks a lot.

LeiWang1999, 2022-02-09:
Clock gating is used to reduce power dissipation, but our FPGA doesn't need ir, and to shutdown the clock gating circuit, you should add the following global definition:

VLIB_BYPASS_POWER_CG
NV_FPGA_FIFOGEN
FIFOGEN_MASTER_CLK_GATING_DISABLED
FPGA
SYNTHESIS

and as descirbed in this post: https://zhuanlan.zhihu.com/p/378202360

Mightyyy, 2022-02-11:
This works. Thank you for your explanation.

closed 2022-02-11 by Mightyyy