LeiWang1999/ZYNQ-NVDLA #15: Mapping the NVDLA accelerator to zynqMP platform

AlanWalker159, 2022-02-20:
非常感谢你分享在zynq700系列上部署NVDLA的具体操作。
麻烦问下，你是否有在zynqMP platform上进行部署呢！
我不知道如何进行IP间的线路连接，麻烦你是否可以分享下相关经验呢？
真诚的感谢。

LeiWang1999, 2022-02-20:
@AlanWalker159 Hi
我现在身边没有Windows的电脑，给不了你连线图，但是在Zynq MPSoc上的部署我是做过的，在这个repo的issue下面也有不少是基于MPSoc来做，如果你有问题兴许可以问问他们？
其实在ZYNQ MpSoc上的连线方式和ZYNQ 7000上基本是一样的，也许这篇post能够帮助到你：https://vvviy.github.io/2018/09/12/nv_small-FPGA-Mapping-Workflow-I/

AlanWalker159, 2022-02-20:
太谢谢您的回复了，我刚刚接触FPGA对IP配置相关的不熟悉，我最近尝试下你提供的连接，然后试试。真诚的感谢。

AlanWalker159, 2022-02-24:
[image]

大佬，还得麻烦请教您一下呢，我这边进行了Ｂlock Design连线后，在综合时出现了截图的错误。Ｖivado工具中也配置了，如下的宏定义的，VLIB_BYPASS_POWER_CG
NV_FPGA_FIFOGEN
FIFOGEN_MASTER_CLK_GATING_DISABLED
FPGA
SYNTHESIS。不知道为什么！太麻烦您了，非常抱歉。

LeiWang1999, 2022-02-24:
@AlanWalker159 你是在Package IP的时候定义的，还是在主工程里面定义的？
应该是Package IP的时候就加上。

AlanWalker159, 2022-02-25:
１. 我在建立工程导入代码时就添加了这些宏定义
2. 打包ＩＰ，保持ＩＰ
3. 新建工程导入ＩＰ，进行block design，这里综合时就出错了，这里也进行了宏定义，也进行了global include file尝试也出错。

AlanWalker159, 2022-02-25:
对了，麻烦请问下您呢，你提供RTL中 vmod下添加了fifo部分，这部分内容好像使用NVDLA官方编译并没有生成呢？

LeiWang1999, 2022-02-25:
我是使用的是hw仓库的master分支的代码，你是用的nvdlav1这个分支的吧？

LeiWang1999, 2022-02-25:
> １. 我在建立工程导入代码时就添加了这些宏定义 2. 打包ＩＰ，保存ＩＰ 3. 新建工程导入ＩＰ，进行block design，这里综合时就出错了，这里也进行了宏定义，也进行了global include file尝试也出错。

@AlanWalker159 两边都加了？那不科学，你可以在打包IP的那个工程综合一下ip，看看会不会报错。

AlanWalker159, 2022-02-25:
> > １. 我在建立工程导入代码时就添加了这些宏定义 2. 打包ＩＰ，保存ＩＰ 3. 新建工程导入ＩＰ，进行block design，这里综合时就出错了，这里也进行了宏定义，也进行了global include file尝试也出错。
> 
> @AlanWalker159 两边都加了？那不科学，你可以在打包IP的那个工程综合一下ip，看看会不会报错。

这边我综合测试了一下的，可以的，没有保持。

LeiWang1999, 2022-02-25:
@AlanWalker159 BD工程里update一下ip应该就好了。

AlanWalker159, 2022-02-25:
> 我是使用的是hw仓库的master分支的代码，你是用的nvdlav1这个分支的吧？

我这边使用的你推荐的docker，我使用里面已存在代码的master分支编译，有fifos的。使用nv_small分支是没有的。

AlanWalker159, 2022-02-25:
> @AlanWalker159 BD工程里update一下ip应该就好了。

这个我就觉得好奇怪的，我觉得也是环境配置有点问题。直接强制修改ip源代码也可以绕过，但是可能不太好。

AlanWalker159, 2022-02-25:
> > @AlanWalker159 BD工程里update一下ip应该就好了。
> 
> 这个我就觉得好奇怪的，我觉得也是环境配置有点问题。直接强制修改ip源代码也可以绕过，但是可能不太好。

对了编译出来的nv_small目录下，spec下的文件需要导入工程么？

AlanWalker159, 2022-02-25:
> @AlanWalker159 BD工程里update一下ip应该就好了。

我在一个工程中打包ＩＰ，ＢＤ　design的，宏定义是global include，貌似还是不行。如下截图：
[image]

LeiWang1999, 2022-02-25:
理论上也行？不需要include吗，我是在Settings里加上的。

AlanWalker159, 2022-03-16:
麻烦请教下，我使用你提供的caffe模型，加载运行内核出现异常，不知道你那边是否有遇到过呢？谢谢麻烦了。
[image]
[image]

ChangAnGuLi, 2022-05-12:
> [image]
> 
> 大佬，还得麻烦请教您一下呢，我这边进行了Ｂlock Design连线后，在综合时出现了截图的错误。Ｖivado工具中也配置了，如下的宏定义的，VLIB_BYPASS_POWER_CG NV_FPGA_FIFOGEN FIFOGEN_MASTER_CLK_GATING_DISABLED FPGA SYNTHESIS。不知道为什么！太麻烦您了，非常抱歉。

请问您是怎么解决这个问题的，可以分享一下吗

AlanWalker159, 2022-06-08:
> > [image]
> > 大佬，还得麻烦请教您一下呢，我这边进行了Ｂlock Design连线后，在综合时出现了截图的错误。Ｖivado工具中也配置了，如下的宏定义的，VLIB_BYPASS_POWER_CG NV_FPGA_FIFOGEN FIFOGEN_MASTER_CLK_GATING_DISABLED FPGA SYNTHESIS。不知道为什么！太麻烦您了，非常抱歉。
> 
> 请问您是怎么解决这个问题的，可以分享一下吗

两种方式可以修改的：
1. 在IP源代码中添加相关的宏
2. 采用global include方式，保证你的include文件是在工程中创建的，不能在外部创建再导入，这个是应该和vivado工具相关，我也觉得很纳闷。

LeiWang1999, 2023-10-19:
close as it has been resolved.

closed 2023-10-19 by LeiWang1999