LeiWang1999/ZYNQ-NVDLA #16: Tengine的运行结果  Segmentation fault

yss-pku, 2022-03-01:
我用的是Tengine给的model_zoo里的   resnet18-cifar10_int8.tmfile  这个文件

运行 `./tm_classification_opendla -m /Tengine/models/resnet18-cifar10-nosoftmax-relu_int8.tmfile -i /Tengine/images/cat.jpg -g 32,32 -s 1,1,1`后

结果
[image]

LeiWang1999, 2022-03-02:
这好像是个bug,,,

yss-pku, 2022-03-03:
> 

好吧，感谢解答！我看你的是直接就打印出了结果？我的前面要打印一大堆东西，要等好久

closed 2022-03-07 by yss-pku

WuZheng326, 2022-03-11:
请问还有后续吗？我也遇到了这个问题，后面有解决吗

3to11CHAINJ, 2022-11-09:
> 我用的是Tengine给的model_zoo里的 resnet18-cifar10_int8.tmfile 这个文件
> 
> 运行 `./tm_classification_opendla -m /Tengine/models/resnet18-cifar10-int8.tmfile -i /Tengine/images/cat.jpg -g 32,32 -s 1,1,1`后
> 
> 结果 [image]

我在推理resnet18-cifar10_int8.tmfile模型的时候也遇到了同样的问题，并且前面打印一大堆信息，请问后续解决了吗。
期待您的回复

WuZheng326, 2022-11-09:
你好，我是吴峥，已收到您的邮件，我会尽快给您回复。