LeiWang1999/ZYNQ-NVDLA #17: NVDLA stuck during inference

suryacharanp, 2022-03-07:
Hi @LeiWang1999 

 I am running the same implementation on ZCU104 with your code files, I insmoded the .ko file generated for MPSoC. I verified the device in "/dev/drm" and interrupt in "/proc/interrupts". They are perfect. 

The problem is that it is getting stuck in the middle of model execution. I am attaching the debug log of KMD

submitting tasks...
[  319.138013] Enter: dla_initiate_processors
[  319.146859] Enter: dla_submit_operation
[  319.150684] Prepare Convolution operation index 0 ROI 0 dep_count 1
[  319.156939] Enter: dla_prepare_operation
[  319.160857] processor:Convolution group:0, rdma_group:0 available
[  319.166939] Enter: dla_read_config
[  319.170342] Exit: dla_read_config
[  319.173649] Exit: dla_prepare_operation status=0
[  319.178259] Enter: dla_program_operation
[  319.182173] Program Convolution operation index 0 ROI 0 Group[0]
[  319.188205] no desc get due to index==-1
[  319.192122] no desc get due to index==-1
[  319.196035] no desc get due to index==-1
[  319.199948] no desc get due to index==-1
[  319.203864] no desc get due to index==-1
[  319.207778] Enter: dla_op_programmed
[  319.211348] Update dependency operation index 3 ROI 0 DEP_COUNT=3
[  319.217431] Update dependency operation index 1 ROI 0 DEP_COUNT=1
[  319.223516] enable SDP in dla_update_dependency as depdency are resolved
[  319.230207] Enter: dla_enable_operation
[  319.234036] exit dla_enable_operation without actual enable due to processor hasn't been programmed
[  319.243071] Exit: dla_enable_operation status=0
[  319.247594] Exit: dla_op_programmed
[  319.251074] Exit: dla_program_operation status=0
[  319.255684] Exit: dla_submit_operation
[  319.259424] Enter: dla_dequeue_operation
[  319.263341] Dequeue op from Convolution processor, index=3 ROI=0
[  319.269337] Enter: dla_submit_operation
[  319.273166] Prepare Convolution operation index 3 ROI 0 dep_count 2
[  319.279423] Enter: dla_prepare_operation
[  319.283340] processor:Convolution group:1, rdma_group:0 available
[  319.289422] Enter: dla_read_config
[  319.292824] Exit: dla_read_config
[  319.296132] Exit: dla_prepare_operation status=0
[  319.300742] Enter: dla_program_operation
[  319.304656] Program Convolution operation index 3 ROI 0 Group[1]
[  319.310685] no desc get due to index==-1
[  319.314601] no desc get due to index==-1
[  319.318519] no desc get due to index==-1
[  319.322432] no desc get due to index==-1
[  319.326347] no desc get due to index==-1
[  319.330261] Enter: dla_op_programmed
[  319.333831] Update dependency operation index 6 ROI 0 DEP_COUNT=3
[  319.339914] Update dependency operation index 4 ROI 0 DEP_COUNT=2
[  319.346000] Exit: dla_op_programmed
[  319.349474] Exit: dla_program_operation status=0
[  319.354080] Exit: dla_submit_operation
[  319.357820] Exit: dla_dequeue_operation
[  319.361649] Enter: dla_submit_operation
[  319.365472] Prepare SDP operation index 1 ROI 0 dep_count 0
[  319.371033] Enter: dla_prepare_operation
[  319.374951] processor:SDP group:0, rdma_group:0 available
[  319.380338] Enter: dla_read_config
[  319.383738] Exit: dla_read_config
[  319.387048] Exit: dla_prepare_operation status=0
[  319.391655] Enter: dla_program_operation
[  319.395571] Program SDP operation index 1 ROI 0 Group[0]
[  319.400888] no desc get due to index==-1
[  319.404806] no desc get due to index==-1
[  319.408722] no desc get due to index==-1
[  319.412635] no desc get due to index==-1
[  319.416549] Enter: dla_op_programmed
[  319.420119] Update dependency operation index 4 ROI 0 DEP_COUNT=1
[... 120 lines cut ...]
 32:         15          0          0          0     GICv2  50 Level     cdns-i2c
 33:          0          0          0          0     GICv2  42 Level     ff960000.memory-controller
 34:          0          0          0          0     GICv2  57 Level     axi-pmon, axi-pmon
 35:          0          0          0          0     GICv2 155 Level     axi-pmon, axi-pmon
 36:         45          0          0          0     GICv2  47 Level     ff0f0000.spi
 37:          0          0          0          0     GICv2  58 Level     ffa60000.rtc
 38:          0          0          0          0     GICv2  59 Level     ffa60000.rtc
 39:          0          0          0          0     GICv2 165 Level     ahci-ceva[fd0c0000.ahci]
 40:       1319          0          0          0     GICv2  81 Level     mmc0
 41:        146          0          0          0     GICv2  53 Level     xuartps
 44:          0          0          0          0     GICv2  84 Edge      ff150000.watchdog
 45:          0          0          0          0     GICv2  88 Level     ams-irq
 46:          0          0          0          0     GICv2 154 Level     fd4c0000.dma
 47:          0          0          0          0     GICv2 151 Level     fd4a0000.zynqmp-display
 48:          0          0          0          0     GICv2 121 Level     a0000000.dla_small_256
 49:          0          0          0          0     GICv2  97 Level     xhci-hcd:usb1
IPI0:      1454       1082       1707       1483       Rescheduling interrupts
IPI1:        17         98        102        136       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
IPI4:       665       2481       2467       2177       Timer broadcast interrupts
IPI5:         0          0          0          0       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts

What do you think the issue is?

LeiWang1999, 2022-03-07:
How about the memory copy sanity in Vivado SDK? Did it work correctly?

suryacharanp, 2022-03-07:
Yes it worked, just now I solved the problem. It is happening because of my lenet loadable, I might have done some mistake in generating it. It is working for the other loadables. Anyway, thank you for responding @LeiWang1999

closed 2022-03-07 by suryacharanp

AlanWalker159, 2022-03-17:
@suryacharanp 
Hi, friend, I meet the same issue as I used the caffe loadable from LeiWang999. Did LeiWang999's loadable work on your platform?

suryacharanp, 2022-04-02:
No, I generated my own loadables. And be careful while doing so, depending on the configuration of the DLA you might need to tweak the compiler code. If it is NVDLA small/large then follow this link[https://github.com/powderluv/nvdla-notes](url)

AlanWalker159, 2022-04-03:
> No, I generated my own loadables. And be careful while doing so, depending on the configuration of the DLA you might need to tweak the compiler code. Follow this link[https://github.com/powderluv/nvdla-notes](url)

Thank you so so much, dear friend. I am going to try again, and I won't give up, :)