LeiWang1999/ZYNQ-NVDLA #18: block design后implementation失败

Aleryaa, 2022-04-28:
您好，我使用ZC706开发板，按照您的博客现在已经完成了block design工作，现在想要继续生成比特流但对bd进行implementation会报错提示IO口数量不足。看您在知乎的回答说综合时的IO超标不影响后续设计，请问这个问题应该如何解决呢？或者说是我在block design的时候犯了什么错误导致了这个问题吗？
[image]
[image]

ChangAnGuLi, 2022-05-11:
我也遇到了同样的问题，请问您解决了吗

Aleryaa, 2022-05-12:
> 我也遇到了同样的问题，请问您解决了吗

我给bd工程create HDL wrapper之后再impl似乎这个问题解决了，但是又遇到了新的问题，很头疼..

ChangAnGuLi, 2022-05-12:
> > 我也遇到了同样的问题，请问您解决了吗
> 
> 我给bd工程create HDL wrapper之后再impl似乎这个问题解决了，但是又遇到了新的问题，很头疼..

我也遇到一个新的问题，您可以帮忙看下
[image]
嘛

Aleryaa, 2022-05-12:
> > > 我也遇到了同样的问题，请问您解决了吗
> > 
> > 
> > 我给bd工程create HDL wrapper之后再impl似乎这个问题解决了，但是又遇到了新的问题，很头疼..
> 
> 我也遇到一个新的问题，您可以帮忙看下 [image] 嘛

我遇到过这个问题，也还没有解决。不过我换了一个环境重新搭了一套工程再impl就变成了新的报错，可能和版本与环境有关系？

ChangAnGuLi, 2022-05-12:
请问您具体是怎么换的环境，我一开始是没有遇到这个问题的并且也是能够顺利输出bit流文件的，但是在现在的windows vivado2019.1环境下就不行了，难道真的要在Linux下复刻作者的步骤吗

ChangAnGuLi, 2022-05-12:
> 请问您具体是怎么换的环境，我一开始是没有遇到这个问题的并且也是能够顺利输出bit流文件的，但是在现在的windows vivado2019.1环境下就不行了，难道真的要在Linux下复刻作者的步骤吗

可能是我电脑的问题在虚拟机环境下impl就很容易闪退卡死

Aleryaa, 2022-05-12:
> 请问您具体是怎么换的环境，我一开始是没有遇到这个问题的并且也是能够顺利输出bit流文件的，但是在现在的windows vivado2019.1环境下就不行了，难道真的要在Linux下复刻作者的步骤吗

我之前是在linux服务器端用vivado 2019.1出现了你这个问题，现在在windows端重新搭了一套，直接使用作者的RTL代码在vivado 2018.3 搭了一个工程，可以成功impl并生成bitstream，但是WNS和TNS为负，不知道会不会有什么问题。

ChangAnGuLi, 2022-05-12:
> 

这样么，我等下试一下，谢谢

ChangAnGuLi, 2022-05-13:
> > 请问您具体是怎么换的环境，我一开始是没有遇到这个问题的并且也是能够顺利输出bit流文件的，但是在现在的windows vivado2019.1环境下就不行了，难道真的要在Linux下复刻作者的步骤吗
> 
> 我之前是在linux服务器端用vivado 2019.1出现了你这个问题，现在在windows端重新搭了一套，直接使用作者的RTL代码在vivado 2018.3 搭了一个工程，可以成功impl并生成bitstream，但是WNS和TNS为负，不知道会不会有什么问题。

我试了好几个vivado版本都不行，您可以分享一下block design文件吗，我对照着看一下，是不是IP配置方面出了问题，我的zynq型号为XC7Z045FFG900-2I

ChangAnGuLi, 2022-05-13:
> 我试了好几个vivado版本都不行，您可以分享一下block design文件吗，我对照着看一下，是不是IP配置方面出了问题，我的zynq型号为XC7Z045FFG900-2I

我修改implementation策略在opt_design中不勾选is_enable，可以正常完成整个implementation，但是在生成bit流文件时报错了

ChangAnGuLi, 2022-05-13:
> 我试了好几个vivado版本都不行，您可以分享一下block design文件吗，我对照着看一下，是不是IP配置方面出了问题，我的zynq型号为XC7Z045FFG900-2I

我在nvdla issue里面找到了解决办法
[image]

LeiWang1999, 2022-05-13:
@ChangAnGuLi 我的代码也是用的master分支的，不然你的硬件会和软件对不上。

Aleryaa, 2022-05-13:
> 我试了好几个vivado版本都不行，您可以分享一下block design文件吗，我对照着看一下，是不是IP配置方面出了问题，我的zynq型号为XC7Z045FFG900-2I

我今天重新试了一下用2019.1搭建工程，成功生成了比特流，意外的WNS和TNS也都是正常值。之前我也都是用的master分支生成的vmod代码，不知道是哪里出了问题...工程文件下载链接：https://drive.google.com/file/d/1Y4c3nsDzgxs-60ai1TgjPhkeOvmQuymj/view?usp=sharing

ChangAnGuLi, 2022-05-13:
> 我今天重新试了一下用2019.1搭建工程，成功生成了比特流，意外的WNS和TNS也都是正常值。之前我也都是用的master分支生成的vmod代码，不知道是哪里出了问题...工程文件下载链接：https://drive.google.com/file/d/1Y4c3nsDzgxs-60ai1TgjPhkeOvmQuymj/view?usp=sharing

谢谢您，对比您的工程文件后我大致知道如何去修改我的设计了

closed 2022-05-14 by Aleryaa