LeiWang1999/ZYNQ-NVDLA #20: implementation error [DRC INBB-3]

ChangAnGuLi, 2022-05-12:
您好，我最近有在向您学习NVDLA移植，开发板型号是XC7Z045FFG900-2I，在implementation时发生了如下错误
[image]
希望您能给出一些修改意见，谢谢

ChangAnGuLi, 2022-05-13:
我在NVDLA issue中找到了一个解决方案
[image]
那我应该怎么找到这vmod呢，用docker镜像生成的难道不是吗，希望您可以指导一下

LeiWang1999, 2022-05-13:
在docker里生成文件，tmake那一步，生成的output文件夹里面的vmod(verlog module)，我是直接用的docker里的，没有用small分支

ChangAnGuLi, 2022-05-13:
> 

我也是按照您的过程做的，但是上述这个问题您有遇到过吗，还是除了在block design中除了连线外还需要别的设置吗，对了，我的clock gating是如下图这样配置的，会不会是这里的问题
[image]

ChangAnGuLi, 2022-05-13:
> 在docker里生成文件，tmake那一步，生成的output文件夹里面的vmod(verlog module)，我是直接用的docker里的，没有用small分支

您有遇到上述问题吗，您能否给出一些修改意见

ChangAnGuLi, 2022-05-16:
问题已经解决

closed 2022-05-16 by ChangAnGuLi

ChangAnGuLi, 2022-05-16:
> 问题已经解决

软件适配问题

H2OD, 2022-10-05:
> > 问题已经解决
> 
> 软件适配问题

想问一下是怎么解决的呀