LeiWang1999/ZYNQ-NVDLA #24: Vivado implementation failed

HanHaoxiang, 2022-10-13:
[image]

[Place 30-58] IO placement is infeasible. Number of unplaced terminals (431) is greater than number of available sites (400).
The following are banks with available pins: 

[Place 30-374] IO placer failed to find a solution
Below is the partial placement that can be analyzed to see if any constraint modifications will make the IO placement problem easier to solve.

+-------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
|                                                                     IO Placement : Bank Stats                                                                           |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
| Id | Pins  | Terms |                               Standards                                |                IDelayCtrls               |  VREF  |  VCCO  |   VR   | DCI |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
|  0 |     0 |     0 |                                                                        |                                          |        |        |        |     |
|  9 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 10 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 11 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 12 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 13 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 33 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 34 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 35 |    50 |     0 |                                                                        |                                          |        |        |        |     |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
|    |   400 |     0 |                                                                        |                                          |        |        |        |     |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+

IO Placement:
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+
| BankId |             Terminal | Standard        | Site                 | Pin                  | Attributes           |
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+

[Place 30-99] Placer failed with error: 'IO Clock Placer failed'
Please review all ERROR, CRITICAL WARNING, and WARNING messages during placement to understand the cause for failure.

[Common 17-69] Command failed: Placer could not place all instances

想请教一下，在按照您的博客笔记执行硬件部分时，实现出现了问题。请问该如何改善，我是应该手动布局布线关掉一些无用的IO吗？

3to11CHAINJ, 2022-10-23:
> [image]
>
> [Place 30-58] IO placement is infeasible. Number of unplaced terminals (431) is greater than number of available sites (400). The following are banks with available pins:
>
> [Place 30-374] IO [...]

我也碰到了相同的问题，请问解决了吗最后

HanHaoxiang, 2022-10-23:
> 问解决了吗最后

将做好的block design文件设置封装，重新设置top文件即可。