LeiWang1999/ZYNQ-NVDLA #27: wrong output.dimg when execute nvdla_runtime

jxyjason, 2023-10-18:
hi, lei
I have followed your blog and finally go to the step of running nvdla_runtime, the log says "test pass" but when I watch the file output.dimg, the result is "12 12 12 12 12 12 12 12 12 12 ", like:
[image]

can you help me, please!

jxyjason, 2023-10-21:
I solve it by modify file libjpeg.h. So I close it.

closed 2023-10-21 by jxyjason

White-Sunday, 2024-01-10:
> I solve it by modify file libjpeg.h. So I close it.
I have encountered the same problem as you. Can you tell me how you modified libjpeg. h?

White-Sunday, 2024-01-10:
> I solve it by modify file libjpeg.h. So I close it.

@jxyjason 
I have encountered the same problem as you. Can you tell me how you modified libjpeg. h?

jxyjason, 2024-01-11:
sure, here is:
in /umd/external/include/jpeglib.h:218, /* Known color spaces. */ part, change is as follows:
`
/* Known color spaces. */

#define JCS_EXTENSIONS 1
#define JCS_ALPHA_EXTENSIONS 1

typedef enum {
        JCS_UNKNOWN,            /* error/unspecified */
        JCS_GRAYSCALE,          /* monochrome */
        JCS_RGB,                /* red/green/blue, standard RGB (sRGB) */
        JCS_YCbCr,              /* Y/Cb/Cr (also known as YUV), standard YCC */
        JCS_CMYK,               /* C/M/Y/K */
        JCS_YCCK,               /* Y/Cb/Cr/K */

        JCS_EXT_RGB,            /* red/green/blue */
        JCS_EXT_RGBX,           /* red/green/blue/x */
        JCS_EXT_BGR,            /* blue/green/red */
        JCS_EXT_BGRX,           /* blue/green/red/x */
        JCS_EXT_XBGR,           /* x/blue/green/red */
        JCS_EXT_XRGB,           /* x/red/green/blue */
        JCS_EXT_RGBA,           /* red/green/blue/alpha */
        JCS_EXT_BGRA,           /* blue/green/red/alpha */
        JCS_EXT_ABGR,           /* alpha/blue/green/red */
        JCS_EXT_ARGB,           /* alpha/red/green/blue */
        JCS_RGB565,              /* 5-bit red/6-bit green/5-bit blue */

        JCS_BG_RGB,             /* big gamut red/green/blue, bg-sRGB */
        JCS_BG_YCC              /* big gamut Y/Cb/Cr, bg-sYCC */
} J_COLOR_SPACE;

/* Supported color transforms. */
//……
`

jxyjason, 2024-01-11:
sure, here is:
in /umd/external/include/jpeglib.h:218, "Known color spaces" part, change is as follows:
```c
/* Known color spaces. */

#define JCS_EXTENSIONS 1
#define JCS_ALPHA_EXTENSIONS 1

typedef enum {
JCS_UNKNOWN, /* error/unspecified /
JCS_GRAYSCALE, / monochrome /
JCS_RGB, / red/green/blue, standard RGB (sRGB) /
JCS_YCbCr, / Y/Cb/Cr (also known as YUV), standard YCC /
JCS_CMYK, / C/M/Y/K /
JCS_YCCK, / Y/Cb/Cr/K */

    JCS_EXT_RGB,            /* red/green/blue */
    JCS_EXT_RGBX,           /* red/green/blue/x */
    JCS_EXT_BGR,            /* blue/green/red */
    JCS_EXT_BGRX,           /* blue/green/red/x */
    JCS_EXT_XBGR,           /* x/blue/green/red */
    JCS_EXT_XRGB,           /* x/red/green/blue */
    JCS_EXT_RGBA,           /* red/green/blue/alpha */
    JCS_EXT_BGRA,           /* blue/green/red/alpha */
    JCS_EXT_ABGR,           /* alpha/blue/green/red */
    JCS_EXT_ARGB,           /* alpha/red/green/blue */
    JCS_RGB565,              /* 5-bit red/6-bit green/5-bit blue */

    JCS_BG_RGB,             /* big gamut red/green/blue, bg-sRGB */
    JCS_BG_YCC              /* big gamut Y/Cb/Cr, bg-sYCC */
} J_COLOR_SPACE;

/* Supported color transforms. */
//……
```

White-Sunday, 2024-01-17:
@jxyjason
Thanks for your help. But after doing so, I ran `make runtime TOOLCHAIN_ PREFIX=/usr/bin/` again, got the following error:
```
DlaImageUtils.cpp: In function ‘NvDlaError JPEG2DIMG(std::__cxx11::string, NvDlaImage*, nvdla::IRuntime::NvDlaTensor*)’:
DlaImageUtils.cpp:200:14: error: ‘JCS_GRAYSCALE’ was not declared in this scope
         case JCS_GRAYSCALE:
              ^
DlaImageUtils.cpp:205:14: error: ‘JCS_YCbCr’ was not declared in this scope
         case JCS_YCbCr: // FIXME: dont know how to handle compression yet
              ^
DlaImageUtils.cpp:206:14: error: ‘JCS_RGB’ was not declared in this scope
         case JCS_RGB:
              ^
DlaImageUtils.cpp:248:14: error: ‘JCS_CMYK’ was not declared in this scope
         case JCS_CMYK:
              ^
DlaImageUtils.cpp:249:14: error: ‘JCS_YCCK’ was not declared in this scope
         case JCS_YCCK:
```

jxyjason, 2024-01-18:
try this again:

/* Known color spaces. */

#define JCS_EXTENSIONS 1
#define JCS_ALPHA_EXTENSIONS 1

typedef enum {
        JCS_UNKNOWN,            /* error/unspecified */
        JCS_GRAYSCALE,          /* monochrome */
        JCS_RGB,                /* red/green/blue, standard RGB (sRGB) */
        JCS_YCbCr,              /* Y/Cb/Cr (also known as YUV), standard YCC */
        JCS_CMYK,               /* C/M/Y/K */
        JCS_YCCK,               /* Y/Cb/Cr/K */

        JCS_EXT_RGB,            /* red/green/blue */
        JCS_EXT_RGBX,           /* red/green/blue/x */
        JCS_EXT_BGR,            /* blue/green/red */
        JCS_EXT_BGRX,           /* blue/green/red/x */
        JCS_EXT_XBGR,           /* x/blue/green/red */
        JCS_EXT_XRGB,           /* x/red/green/blue */
        JCS_EXT_RGBA,           /* red/green/blue/alpha */
        JCS_EXT_BGRA,           /* blue/green/red/alpha */
        JCS_EXT_ABGR,           /* alpha/blue/green/red */
        JCS_EXT_ARGB,           /* alpha/red/green/blue */
        JCS_RGB565,              /* 5-bit red/6-bit green/5-bit blue */

        JCS_BG_RGB,             /* big gamut red/green/blue, bg-sRGB */
        JCS_BG_YCC              /* big gamut Y/Cb/Cr, bg-sYCC */
} J_COLOR_SPACE;

/* Supported color transforms. */
//……

jxyjason, 2024-01-18:
try this again:

```c
/* Known color spaces. */

#define JCS_EXTENSIONS 1
#define JCS_ALPHA_EXTENSIONS 1

typedef enum {
        JCS_UNKNOWN,            /* error/unspecified */
        JCS_GRAYSCALE,          /* monochrome */
        JCS_RGB,                /* red/green/blue, standard RGB (sRGB) */
        JCS_YCbCr,              /* Y/Cb/Cr (also known as YUV), standard YCC */
        JCS_CMYK,               /* C/M/Y/K */
        JCS_YCCK,               /* Y/Cb/Cr/K */

        JCS_EXT_RGB,            /* red/green/blue */
        JCS_EXT_RGBX,           /* red/green/blue/x */
        JCS_EXT_BGR,            /* blue/green/red */
        JCS_EXT_BGRX,           /* blue/green/red/x */
        JCS_EXT_XBGR,           /* x/blue/green/red */
        JCS_EXT_XRGB,           /* x/red/green/blue */
        JCS_EXT_RGBA,           /* red/green/blue/alpha */
        JCS_EXT_BGRA,           /* blue/green/red/alpha */
        JCS_EXT_ABGR,           /* alpha/blue/green/red */
        JCS_EXT_ARGB,           /* alpha/red/green/blue */
        JCS_RGB565,              /* 5-bit red/6-bit green/5-bit blue */

        JCS_BG_RGB,             /* big gamut red/green/blue, bg-sRGB */
        JCS_BG_YCC              /* big gamut Y/Cb/Cr, bg-sYCC */
} J_COLOR_SPACE;

/* Supported color transforms. */
//……
```

White-Sunday, 2024-01-24:
@jxyjason
I have tried this method on the jpeg-6, jpeg-9, and jpeg-9e libraries, but it did not work. In addition, I also attempted to compile jpeg-6b and jpeg-6b2, but both ended in failure.

I printed the input buffer through `tensorImage->printBuffer(true);` in `umd\apps\runtime\RuntimeTest.cpp` line 106. As shown in the figure below, the problem may occur after the task is submitted.
[image]

我正在进行nvdla有关的毕业设计，想了解更多的细节。如果方便的话，希望加个微信交流一下。这是我的微信：wanSD777。
再次感谢你的帮助。

jxyjason, 2024-01-25:
sure

White-Sunday, 2024-02-14:
问题解决了。

整个网络的最后一层（softmax层）是交由模拟器，也就是CPU来处理的。驱动需要将加速器写到reserved-memory的数据mmap到内核态/用户态的虚拟地址空间，供上层应用访问。将softmax层的输入dump出来后结果都是0，正好对应了“12 12 12 ...”的结果。

这里就引出了两个问题：

**第一个问题**：32 位arm linux允许分配的vmalloc空间最大是240MB，预留256MB的DMA地址空间会导致注册驱动时报错：

[image]

可以通过将DMA地址空间替换成128MB来解决这个问题：

修改system-user.dtsi成如下内容：

```
/include/ "system-conf.dtsi"
/ {
   reserved-memory {
      #address-cells = <1>;
      #size-cells = <1>;
      ranges;

      nvdla_reserved: buffer@0x38000000 {
         compatible = "shared-dma-pool";
         no-map;
         reg = <0x38000000 0x08000000>;
      };
   };
};

&NV_nvdla_wrapper_0{
    compatible = "nvidia,nv_small";
    memory-region = <&nvdla_reserved>;
};
```

对应对应修改一下nvdla_gem.c的内容（4.19内核里dma_declare_coherent_memory返回0为success）：

```c
dma = dma_declare_coherent_memory(drm->dev, 0x38000000, 0x38000000,
                                      0x08000000, DMA_MEMORY_EXCLUSIVE);
    if (dma) {
        err = -ENOMEM;
        goto unref;
    }
```

**第二个问题**：AXI总线封装问题。

参考博客在做IP Package时，为了与AXI总线对齐而添加的多余信号（`m_axi_*`），在我的板卡（AX7Z100）上，影响了AXI SmartConnect这个IP核的正常工作。

[image]

带着`m_axi_*`信号在block design里进行validate design时会有个警告：

```
CRITICAL WARNING: [xilinx.com:ip:smartconnect:1.0-1] design_1_smartconnect_0_0: RUSER_WIDTH (1) of S00_AXI must be integer number of bits per byte of RDATA (64).
CRITICAL WARNING: [xilinx.com:ip:smartconnect:1.0-1] design_1_smartconnect_0_0: WUSER_WIDTH (1) of S00_AXI must be integer number of bits per byte of WDATA (64).
```

结果会导致总线无法正常工作，dump出来的数据都是0。

解决方法为不用添加多余的`m_axi_*`信号，因为AXI SmartConnect这个IP核会自动帮助你处理总线映射。