LeiWang1999/ZYNQ-NVDLA #29: Package IP

priorityqueueCarl, 2025-02-17:
I automatically deduced the encapsulation of the APB as follows:

[image]

However, when I automatically derive the package AXI4, I can't find the nvdla_coredbb* related parts, how can I automatically derive the package AXI4?

LeiWang1999, 2025-02-17:
Checkout the release https://github.com/LeiWang1999/ZYNQ-NVDLA/releases/tag/v0.1
I've provided a video script to help you release and trace bugs.

priorityqueueCarl, 2025-02-17:
> Checkout the release https://github.com/LeiWang1999/ZYNQ-NVDLA/releases/tag/v0.1 I've provided a video script to help you release and trace bugs.

Thank you very much for your help!

closed 2025-02-17 by LeiWang1999